MiSTle-Dev/NanoMig #3: graphics corrupted with more than 512k chipmem

mibosshard, 2024-06-07:
tested with Kickstart 1.3 and 2.05: as soon as more than 512k of chipmem is activated, graphics glitches appear (e.g. when resizing or moving windows in Workbench) or the system hangs (e.g. running boing-ball demo).

running NanoMig with 512k of chipmem only, the system seems to be rock solid, AFAICT.

- switching PAL/NTSC has no effect
- switching OCS/ECS has no effect
- switching on/off or resizing slowmem has no effect

harbaum, 2024-06-07:
The whole system is unstable and many games and demos don't run properly. This is well-known.

harbaum, 2024-06-23:
I've had a closer look at the ram timing and I'd say that the CPU can access ram pretty reliably.

I assume it's one of the dma units that has problems accessing the ram. This may also explain other graphical glitches I see in demos.

I am no Amiga user. But I think I've read somewhere that the decoding of mfm floppy data is done by the blitter. That could also explain why floppy I/o seems to have issues as the blitter imho also uses dma.

But as I said, I basically know nothing about amigas ...

darius600, 2024-06-24:
What I read Blitter can be used for MFM decoding: https://en.wikipedia.org/wiki/Amiga_Original_Chip_Set#Floppy_disk_controller
Or it can also be done with CPU only. Here 2 Trackloaders without Blitter: 
https://github.com/deplinenoise/trackloader/blob/master/bootblock.asm
https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/framework/loader.asm 

What I see Blitter.v in Nanomig is (almost) identical with Blitter.v in minimig-de1:
https://github.com/rkrajnc/minimig-de1/blob/master/rtl/minimig/Blitter.v (master)
https://github.com/rkrajnc/minimig-de1/blob/4677847cfe7b7cb28066d980d687b2c158d037ab/rtl/minimig/Blitter.v (the rel 4 I am using, which is working fine)

JJDSNT, 2024-07-12:
just out of curiosity, how does aros rom behaved?

https://github.com/aros-development-team/AROS/blob/master/boot/modular_kickstart.txt

harbaum, 2024-08-20:
Does this problem still exist?

mibosshard, 2024-08-20:
> Does this problem still exist?

I really can't say... the latest release seems to be hit and miss and it randomly starts, or not. I also reflashed the Tang Nano 20k with the latest MiSTeryNano just to make sure it's not my hardware and it ran flawless.

NanoMig v0.3 (tested with Kickstart 1.3 Rev 34.005 flashed at offset 0x400000 and 0x440000)

I did 30x power cycles (wait a few seconds before powering it on again):
black screen: 8x (pressing S2 does nothing, power cycle needed to bring it back to life)
Kickstart screen OK (Workbench will boot) 9x
garbled Kickstart screen (Workbench will not boot) 13x

I did 30x reset cycles (press S2)
black screen: -
Kickstart screen OK (Workbench will boot) 12x
garbled Kickstart screen (Workbench will not boot) 18x

to me it looks like it doesn't matter if I press S2 or initiate a reset from the OSD menu.
I also tried different µSD cards (all of them being premium brands): AFAICT, this has no effect.

regards,
michael

harbaum, 2024-08-20:
Yes, start reliability is not that good. Although IMHO not as bad as you describe. Perhaps some timing close to the limit which affects so e devices more than others. I have changes the reset timing lately and it seems to improve.

mibosshard, 2024-08-20:
if the system's booting up... NanoMig v0.3 appears to work just fine if configured with more than 512K of chipmem. so that's good progress.

harbaum, 2024-09-03:
Seems to be fixed