MiSTle-Dev/NanoMig #4: stuttering mouse pointer

mibosshard, 2024-06-07:
the system's overwhelmed by the amount of input during movement of the mouse. input frequency should be cut in half?

harbaum, 2024-06-07:
This is a general issue and doesn't only affect the NanoMig. This will be solved in the firmware.

hugoferreira, 2024-06-14:
Hey @harbaum! Thank you for all the effort in porting minimig to this little board. May you provide a little bit more information on the M0S firmware, including its source code? May I assume it's the same one used for [MiSTeryNano](https://github.com/harbaum/MiSTeryNano/tree/main)? Thx!

harbaum, 2024-06-15:
> Can you provide a little bit more information on the M0S firmware, including its source code?

@hugoferreira it's indeed the firmware from the MiSTeryNano and the Amiga specific code is included there as well. The SPI documentation from there applies as well.

I am currently rewriting the firmware to make it also compatible with the raspberry pi pico and perhaps the esp32 since the bouffalo SDK for the m0s/bl616 is a constant pain. The pico port will e.g. use the tinyUSB stack which seems more mature than bouffalos CherryUSB.

It's actually a pity since the hardware of the m0s still seems pretty powerful and neither the rp2040 nor the esp32 are IMHO on par.

hugoferreira, 2024-06-16:
> It's actually a pity since the hardware of the m0s still seems pretty powerful and neither the rp2040 nor the esp32 are IMHO on par.

I hear you. As always, the absence of a good software stack ruins the hardware. 

The ESP32 might be a good alternative for wireless keyboard, mouse and gamepad. Actually, using it for the joystick is relatively trivial, given how the pins are break out. Is there any documentation on the SPI interface you use for keyboard and mouse comms to the FPGA? I might take a stab at it.

harbaum, 2024-06-23:
The SPI interface is documented at https://github.com/harbaum/MiSTeryNano/blob/main/SPI.md

This has already been ported to the rp2040 at https://github.com/harbaum/FPGA-Companion

I am currently already working on the ESP32-S2/S3 support as These have (rather limited) USB host support and would probably make most sense in combination with wireless input devices.

closed 2026-10-22 by Manger74