MiSTle-Dev/NanoMig #5: TF card preferences?

BennehBoy, 2024-06-12:
I can't get any adf to load when I place it on a variety of different TF cards with the name df0.adf

Have I missed anything?  Are there any recommendations for card brand and size.  All of the cards I've tried work fine with nestang, but I realise this could be down to differences in driver implementations..

harbaum, 2024-06-12:
As written in the main README, this is a very early test release. It's not supposed to work for anyone, but instead proves that a Minimig on Tang Nano is possible at all. It's now up to some exüperienced Amiga users (which I am not) to make this into a working thing ... or not.

harbaum, 2024-06-12:
All cards I've used so far have been working. This includes some no name 16GB cards from far east as well as high endurance 32GB ones from Sandisk.

But I doubt it's the SD card itself you are having trouble with. For the SD card to work all components incl. the M0S have to work properly.

BennehBoy, 2024-06-12:
OK, so now I know the obvious issue, I have not connected an M0S as I'm waiting on the to ship.  Apologies for wasting your time.

harbaum, 2024-06-12:
The M0S is responsible for the SD cards own file system. It basically makes the stored disk files accessible to the core.

darius600, 2024-06-18:
I was puzzled by the same thing as @BennehBoy 
My initial expectation was that even without the M0S dock the NanoMig would start.
I propose to update the README (see PR above) and close this issue.

kragil, 2025-01-18:
Sorry to resurrect this old bug, but I still have a question and I think this is the right issue to ask. 
I was under the (probably really naive) impression that I could build an Amiga with an old Amiga joystick ( https://github.com/harbaum/NanoMig/issues/26 ) with just the Nano 20k and nothing else. That does not seem to work.

Could you explain why the M0S or a Pi Pico (I have one of those at work) is needed to be able to read from SD-card? That is really not clear to me.  
Is there a workaround to get just one disk into the Nano without the SD-card? (Tbh I just want to play Turrican 2 ;-) ).

sbeckman, 2026-02-15:
Note that you have to go into the on-screen menu (accessed with the F12 key on your keyboard) and select a floppy disk image to be loaded into the floppy drive for booting. Otherwise the Amiga will stay at the boot screen and wait for a bootable floppy disk to be inserted.