MiSTle-Dev/NanoMig #7: RSI megademo crash during decryption

darius600, 2024-06-18:
The [RSI megademo](https://www.pouet.net/prod.php?which=3119 ) crashes during the decryption of the second part. 
Some people in the comments on pouet mention similar issues in WinUAE related to floppy drive emulation speed.

This issue does not occur with [minimig-de1 rel4](https://github.com/rkrajnc/minimig-de1/releases/tag/minimig-de1-fpga-rel-4) on an Altera DE1. Rel4 is non-AGA, non-OR1200 source from 2012.

harbaum, 2024-06-18:
The whole setup isn't stable and needs some work by someone who knows and loves the Amiga. I have never owned one mysef and haven't even used a real one at all. This core is a proof-of-concept to get some amiga anthusiast started. But it's not a working core.

BennehBoy, 2024-06-19:
Could you link the exact source repo that you used for the minimig basis? I can see comments from the original authot in your repo that are newer than the most recent commits in the google source repo... maybe I just haven't found the correct original source...

BennehBoy, 2024-06-19:
> Could you link the exact source repo that you used for the minimig basis? I can see comments from the original author in your repo that are newer than the most recent commits in the google source repo... maybe I just haven't found the correct original source...

Never mind, it helps if I RTFM -> https://github.com/emard/Minimig_ECS/blob/master/Minimig_ECS_Diamond_Project.zip

darius600, 2024-06-19:
Also see [this chart](https://retroramblings.net/downloads/MinimigFamilyTree.pdf) for the history of Minimig development.
emard/Minimig_ECS is based on Minimig for FleaFPGA Ohm which is mentioned in the above chart.

darius600, 2024-07-22:
The second parts loads in the Verilator simulation.

[image]

Also: In FPGA the boot logos of Red Sector and TCC Design have some visual artifacts (some black pixels are not black). This does not happen in simulation, no artifacts.

harbaum, 2024-07-22:
There are a few things that are odd about the problems. One is that they are quite stable. It's not this typical build-to-build instability with the FPGA just being run close at its limits somewhere. Instead the problems always occur at the same places in a simular manner.

Also it's not DMA in general that has a problem. It seems video and copper run quite stable and also many software just loads fine from floppy. So floppy DMA often just works as expected and it's always the same software that fails to load properly.

For me still GOA and the like don't work under Linux. So I currently have no way to analyze the running FPGA ... not that I have any idea what exactly to look for if I had the possibility.

I have messed a little bit with the code and I think I can rule out a few things. E.g. I can do major changes to SDRAM implementation without affecting the symptoms. So I don't think it's the SDRAM being driven incorrectly. Also classic RAM tests e.g. in DiagROM don't see any issues with ram.

darius600, 2024-08-18:
No issues anymore with the code based on MiSTer.

closed 2024-08-18 by darius600