MiSTle-Dev/NanoMig #8: Planet-Rocklobster - crash during loading

darius600, 2024-06-25:
Please don’t worry I won’t post every demo that crashes. :)

However this one might give a hint since the source code of the loader is available here:
https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/framework/loader.asm 
(no Blitter used)

Status:
- Winuae 5.3 (OCS, Floppy Drive Emulation Speed: 100, 200, 400%, 800%): Working
- Winuae 5.3 (OCS, Floppy Drive Emulation Speed: Turbo): Immediate Crash (before Green Intro/Loader Screen) with Guru Meditation Error: 8000 0003 Task: 00C034F8
- Minimig-de1 rel4 (OCS, Floppy Speed: 1x and 2x): Working
- Nanomig (OCS, Floppy Speed: Normal): Green Intro/Loader Screen appears, crash before start of next part with Guru Meditation Error: 8000 0003 Task: 00C034F8

In the loader code there are some busy waits, e.g.:
btst #1,$1f(a6) ; wait until data read

And scan-line-based waiting using the LdrWait function

darius600, 2024-07-18:
I've modified [`loader.asm`](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/framework/loader.asm) and built a minimum version of the demo. This was very straightforward, thanks to efforts of Axis. 

The crash does not seem to be caused by the waiting periods. I inserted additional wait instructions, but the crash still occurs.

### Modified Code Snippet:

```asm
LdrWait:					
    lsl.w   #2, d6         ; Shift d6 left by 2 bits (multiply by 4)
.loop1:

.Wrdy:	
    btst    #1, $1f(a6)    ; Wait until data is read
    beq.s   .Wrdy
    ; Introduce a brief delay using LdrWait
    move.l  d6, -(sp)      ; Back up d6 before using it for delay
    move.w  #2000, d6      ; Set the amount of scanlines to wait (2000 = 128 ms)
    bsr     LdrWait        ; Call the LdrWait function
    move.l  (sp)+, d6

.DiskR:	
    btst    #5, $f01(a4)   ; Wait until motor is running
    bne.s   .DiskR
    ; Introduce a brief delay using LdrWait
    move.l  d6, -(sp)      ; Back up d6 before using it for delay
    move.w  #2000, d6      ; Set the amount of scanlines to wait (2000 = 128 ms)
    bsr     LdrWait        ; Call the LdrWait function
    move.l  (sp)+, d6
```

harbaum, 2024-07-18:
With

https://github.com/harbaum/NanoMig/tree/main/sim%2Ftest_rom

It should be possible to build something that exposes this problem in the simulation.

harbaum, 2024-07-19:
Can you give us an ADF that crashes as early as possible?

harbaum, 2024-07-19:
Ah ... same red screen with 512k RAM on FPGA ... will need to update the simulation to 1MB ...

darius600, 2024-07-19:
> Ah ... same red screen with 512k RAM on FPGA ... will need to update the simulation to 1MB ...

Yes thats it.
This is the source of the bootblock: https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/launcher/bootblock.asm
Line 187 is the reason for the color. There are 3 paths to "error" and I've just removed the other two.

harbaum, 2024-07-19:
Just updated the simulation. The bad news: The demo does not crash in simulation

[image]

harbaum, 2024-07-19:
This rules out the CPU integration and any fundamental logical problem. 

Not part of simulation is the SD card which I don't expect to be an issue as the Atari ST core used the identical implementation just fine. The SD-RAM is still a possible candidate ...

darius600, 2024-07-19:
The starfield is quite fast actually :)

> Can you give us an ADF that crashes as early as possible?
I have some variants I will test in FPGA to see if some of them don't crash.

harbaum, 2024-07-19:
I don't think testing other variants will give more information. I will have a closer look at the sdram.

darius600, 2024-07-21:
I created some variants, all of which work in WinUAE. **One of them works on FPGA**.

### Variants:
1. **No music replay**: Crashes after boot fadeout (as usual)
2. **No music depack**: Crashes before/during boot fadeout
3. **No music linked**: Crashes after boot fadeout, which is very, very slow, the animation takes ~1 minute
4. **No music linked, shorter text on boot screen**: Starfield starts

### Interpretation
It could be that the next part overwrites parts of the current part, possibly the end. So the current part cannot return properly.

### Details
**Load Order**: [Line 6 of script.asm](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/launcher/script.asm#L6)
1. Green boot screen
2. Music load and start
3. Starfield

**No music replay / No music depack**: [Lines 35 / 40 of launcher_framework.asm](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/launcher/launcher_framework.asm#L35)

**Linking of music**: [Line 121 of build.bat](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/build/build.bat#L121)

**Text of boot screen**: [Line 223 of boot.asm](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/boot/boot.asm#L223)

harbaum, 2024-07-21:
Yes, there are some hints that something is overwritten at floppy load. E.g. the DiagROM also crashes after running floppy IO tests. This of course points to the heavily modified floppy routines. But the part sending the data via DMA is basically untouched. This could explain wrong data. But I wouldn't understand why this might overwrite something.

Also some demos load just fine and still have graphical issues (e.g. Arte). A very interesting one is imho the basic demo from the workbench disks. Without knowing the Amiga very good, I think I see aprts of the line graphics being drawn outside the window which would match "data being written where it shouldn't" but in this case by blitter and not dma.

darius600, 2024-08-05:
### TL;DR
It appears that track data is read correctly from disk, but CHIPMEM is getting corrupted by unknown means.

### Details

I have modified [`loader.asm`](https://github.com/AxisOxy/Planet-Rocklobster/blob/master/demo/framework/loader.asm#L232) to output debug messages and the contents of the buffer `ldr_TrackBuffer` on the screen. 
`ldr_TrackBuffer` contains the data after it has been loaded from the disk and **after** MFM decoding was performed by the loader.

Chunks of 256 bytes are output on the screen, as shown in the example below.
[image]

This results in approximately 300 screenshots with TrackBuffer data from WinUAE and Nanomig each.

A visual comparison using OpenCV ([tutorial](https://docs.opencv.org/4.x/d7/d4d/tutorial_py_thresholding.html)) revealed no differences in the images containing data from the buffer. 
However, differences were detected in images containing memory addresses, which I consider an indicator that the difference detection works as needed.

As before, the starfield part successfully starts in WinUAE, while Nanomig crashes after loading. 
I also observed corruption of data in CHIPMEM: Parts of the font data and the constant debug messages are slightly corrupted.

- `ldr_TrackBuffer` is located in FASTMEM.
- The code and data (e.g. debug messages and font data) of the loader is located in CHIPMEM.

Any ideas what to try next to gain more information?

harbaum, 2024-08-07:
According to https://www.atari-forum.com/viewtopic.php?p=467432#p467432 GitHub user @retrofun has managed to get GAO running under Linux. So he's able to track live signals in the running FPGA.

Is the corrupted memory random? Or is there a pattern in the corruption (e.g. a certain address range or a certain data)? If yes, then GAO could be used to trigger onto this pattern to finally figure out when and how the memory is being corrupted.

darius600, 2024-08-09:
> Can you do a rom ckecksum before and after this? The ROM (hopefully) cannot be written by the amiga itself at all but actually resides in ram (being copied there from the flash which is too slow). If the ROM also gets currupted this may actually indicate that refresh is an issue or that ram addressing is broken ...

Done, so far the checksums have been identical.

> According to https://www.atari-forum.com/viewtopic.php?p=467432#p467432 GitHub user @retrofun has managed to get GAO running under Linux. So he's able to track live signals in the running FPGA.

Based on his comment I got it working on Windows.

> Is the corrupted memory random? Or is there a pattern in the corruption (e.g. a certain address range or a certain data)? If yes, then GAO could be used to trigger onto this pattern to finally figure out when and how the memory is being corrupted.

I will try to trigger on the font data which is being corrupted. 
Attached is the output currently captured by GAO when writing MFM decoded data into the buffer: [nanomig_full_core0.zip](https://github.com/user-attachments/files/16567527/nanomig_full_core0.zip)
Do you recommend to add/remove anything?

harbaum, 2024-08-10:
I'll have a look. I am currently restarting from scratch with the mister code. I don't know if it will even fit even if I disable many parts. Our current issues may also just persist or reappear. So it definitely makes sense to understand the problem of the current version.

And it seems you guys are pretty close to being able to track down the issue. Let's hope it's not a fundamental thing like the fpga being just too slow.

retrofun, 2024-08-10:
Hello!

Unfortunately I cannot help further currently because I don't have a m0s dock yet. :slightly_frowning_face:
The seller from which I ordered one sent back my money - reason: he can't ship to me.
I have ordered another m0s dock from a different seller, estimated delivery date is end of August.
Let's hope that works...

harbaum, 2024-08-10:
Do you have a pi pico? That can in the meantime be used as well. Or a esp32-s2 or s3

retrofun, 2024-08-11:
FPGA-Companion... I remember you mentioned something about porting the firmware to other microcontrollers.
Found it in the [MiSTery on Tang Nano](https://www.atari-forum.com/viewtopic.php?p=465372#p465372) thread.

A Pi Pico should be here next Wednesday ;)

In the meantime I have successfully compiled the `fpga_companion.uf2` binary in a Fedora 40 container.
This needed some fixes that I have submitted in a PR.

darius600, 2024-08-11:
I tried triggering on sd_cmd == 4 and the memory region of the font data. 
A match only occurs at the very start - I guess when the program is loaded into memory.
Logs attached.
[nanomig_full_core0.zip](https://github.com/user-attachments/files/16576246/nanomig_full_core0.zip)

darius600, 2024-08-13:
I've applied the checksum (created for the kickstart mem area, see above) to the memory area containing the font data. 
Besides the visual differences (which are quite evident in this version), the checksum also **detects the change**.

Checksum 1
[image]
Checksum 2 (ignore the blue background)
[image]

In **WinUAE**, there are no visual artifacts and the **checksum remains identical**.

Based on this, I'd drawn the following conclusions:

- checksum calculation is functioning correctly (note that no changes to kickstart have been detected earlier).
- font data in memory is getting corrupted and the visual artifacts do not originate from the bitplane.

harbaum, 2024-08-13:
Excellent ... now we "just" need to understand when exactly the ram is corrupted.

harbaum, 2024-08-13:
Btw: the fact that rom (also stored in ram at runtime) stays intact tells us that this e.g. is not a problem with ram refresh are the addressing totally screwed up. This in turn means that there should be an actual ram write which should not happen at all.

I am very curious which part does that. I don't think it's the CPU itself. I'd still expect it to be the blitter and/or another dma unit.

darius600, 2024-08-14:
I am currently unable to detect the ram write that corrupts the data. This is my config. Font data starts at 00C08BAA.
[image]

What I am detecting are ram writes in the very beginning. I would expect this is the initial loading of the program into memory. At this point font data is not corrupted.
[image]

Also interesting: This is slow ram getting corrupted. I do not know about Minimig but OCS should not be able to access this memory region.

harbaum, 2024-08-15:
If slow ram is enabled the Minimig can IMHO access it in any configuration. This should actually pretty easy to verify. Just set the machine type to OCS, enabled 512k chip and 512k slow mem. Boot into workbench and if you see more than 512k bytes being free, then the AmigaOS has seen the slow mem.

So you don't see any ram access to the area but the contents get corrupted? Hmmm ...

I am still in the progress of porting Minimig-Mister-AGA. Simulation works and today I was able to run a first few tests on the FPGA. The errors have changed and e.g. Rocklobster now runs further but then it also reliably hangs with some graphics error. It looks like the error pattern has changed slightly but the general error is still there. Changes are that i'll join your efforts using GAO ...

harbaum, 2024-08-15:
Rocklobster now runs much further .. had to enable some additional ram.

amiga23, 2024-08-15:
@harbaum it's correct, that with OCS an Amiga can have 512k chip and 512k slow ram, but from my understanding the OCS chipset (Agnus) is unable to access anything above the 512k chip RAM. Maybe there is the issue? The Blitter should not be able to access the slow ram. Only the CPU can access slow RAM, but not the Chipset (Paula, Agnus, Denise). Slow RAM should be mapped to $00C00000.

harbaum, 2024-08-15:
I would be surprised if Minimig would allow dma access to slowmem where the real Amiga wouldn't. The nanomig touches none of this so this would be present in all minimig versions ... Which I doubt.

Anyways, I am only a few days from updating the minimig code base. Once that's done I'll have a closer look at the remaining problems.

darius600, 2024-08-16:
> So you don't see any ram access to the area but the contents get corrupted? Hmmm ...

Yes. This could of course also be a configuration issue on my side.
In the meantime, I also tried to match on we_R instead of sd_cmd with the same result.

> I would be surprised if Minimig would allow dma access to slowmem where the real Amiga wouldn't. The nanomig touches none of this so this would be present in all minimig versions ... Which I doubt.

Exactly. In the OCS configuration there won't be (at least should not be) DMA by the Amiga customs chips to the memory region that gets corrupted.

There could be memory access by the CPU (which should be detectable).
Or corruption is caused by the SDRAM itself.

I did some minor tests with the SDRAM controller (changed wait times, cleared we_R and sd_addr) but no luck so far. Do you have a recommendation how to speed up development on this part? Currently I do a code change, synthesize it, flash it (sometimes multiple times are needed) -> 5-10 minutes wait time per code change.

harbaum, 2024-08-16:
The development cycles when doing this kind of stuff are that long. That's why I try to do as much development in the simulation as possible.

But ... the mister code looks quite promising by now. I can run all sanity demos just fine and also planet rocklobster runs flawlessly now. There are a few minor things to iron out as e.g. the joystick isn't working, yet and the system won't boot reliably. But once it's running it seems to be a major improvement over the current version.

pscheir, 2024-08-16:
> The development cycles when doing this kind of stuff are that long. That's why I try to do as much development in the simulation as possible.

OK thanks for the confirmation.

> But ... the mister code looks quite promising by now. I can run all sanity demos just fine and also planet rocklobster runs flawlessly now. There are a few minor things to iron out as e.g. the [...]

Awesome! I am really looking forward to it! 
This means the core will even have AGA?

darius600, 2024-08-16:
> The development cycles when doing this kind of stuff are that long. That's why I try to do as much development in the simulation as possible.

OK thanks for the confirmation
 
> But ... the mister code looks quite promising by now. I can run all sanity demos just fine and also planet rocklobster runs flawlessly now. There are a few minor things to iron out as e.g. the [...]

Awesome! I am really looking forward to it!
This means the core will even have AGA?

harbaum, 2024-08-16:
I could probably enable AGA. Didn't try that. But AGA without 68020, fast ram and caches is probably pretty useless. Also AGA needs faster/wider chip ram interface which NanoMig currently doesn't have.

JJDSNT, 2024-08-16:
hi, i am just curious of what will be possible or not, so i was comparing the files from one repo to another, will nanomig has hrtmon?

(i had lots of curiosity what will be possible with nano but i will not bother beforehand as i am allready happy with the new code base as i think, not sure, it is the reference of implementantion of the minimig core)

Congrats! :)

harbaum, 2024-08-16:
You can add, extend .... whatever you want. You can add/enable AGA, 68020, rtg, toccata, hrtmon, you name it. For now most of this is disabled as my current goal is to get a basic Amiga 500 like device running stable. But the code still exists. 

I am still no Amiga user, have never owned one and don't even know what hrtmon is. If it's useful, then someone might add/reenable it.

It actually depends on you even if you are no coder. Spread the word, get people involved and everything is possible.

There are much bigger Tang FPGA boards, there are some with LCD screens built-in, some with ethernet, there's a carrier board with physical MIDI to make use of ... there's so much to explore ...

retrofun, 2024-08-17:
Since Wednesday I have Nanomig running on the Tang Nano 20K with a Raspberry Pi Pico as FPGA companion.
Setup was easy and it immediately worked! Thanks Till for the good documentation of the project(s)!

----

I can confirm
* that the ROM area $f80000-$ffffff is not writeable
* that there is memory corruption. This also happens with 512KB chipmem only (reproduceable with this [intro](http://janeway.exotica.org.uk/release.php?id=6291). Single pixels in the text get corrupted/are transparent.).

From the comments it looks like with the switch to the Minimig-Mister-AGA the corruption issue should be gone.

----

Regarding memory configuration: I noticed is that memory configuration selected from OSD is not applied. This is a known issue/limitation of Kickstart 1.2/1.3 that do not support "hot plugging" of memory because of "caching"/[a quick setup routine on soft reset at $fc014c](https://wandel.ca/homepage/execdis/exec_disassembly.txt).

This can be confusing and is inconvenient.

To reproduce:

Start Nanomig
- with default configuration 512KB chip/512KB slow memory. Boot Workbench/type _avail_ in CLI/Shell: memory configuration is 512KB/512KB.
- change memory to e.g. 1MB chip/1MB slow memory (Nanomig is reset when changing memory - this is good!). Boot Workbench or type _avail_ in CLI/Shell: memory configuration is still 512KB/512KB.

In MiST firmware we have [a patch](https://github.com/mist-devel/mist-firmware/blob/master/fpga.c#L505) for this.

I've implemented it in the copy routine of Kickstart from flash memory to ROM area in memory:
```
--- a/src/tang/nano20k/top.sv
+++ b/src/tang/nano20k/top.sv
@@ -608,10 +610,15 @@ always @(posedge clk_28m or negedge mem_ready) begin
                 flash_addr <= flash_addr + 22'd1;
                 word_count <= word_count - 22'd1;
 
-                // we don't necessarily need to latch the data. But latching it here
-                // allows to exactly determine the real access time by adjusting flash_cnt
-                // to the lowest value that gives a stable image
-                flash_doutD <= flash_dout;
+                if ((flash_addr == 22'h2000aa || flash_addr == 22'h2200aa) && flash_dout == 16'h6678)
+                    // transform bne.b to bra.b in Kickstart ROM 1.2/1.3 @ $f80154 (mirror) and $fc0154
+                    // this forces memory detection on every reset
+                    flash_doutD <= flash_dout & 16'hf0ff;
+                else
+                    // we don't necessarily need to latch the data. But latching it here
+                    // allows to exactly determine the real access time by adjusting flash_cnt
+                    // to the lowest value that gives a stable image
+                    flash_doutD <= flash_dout;
             end
         end
```

harbaum, 2024-08-17:
Fixed :-)

closed 2024-08-17 by harbaum