MiSTle-Dev/NanoMig #9: VATestprogram: Screen (including NanoMig menu) rolls vertically

darius600, 2024-07-18:
Steps to reproduce:

- Download https://aminet.net/package/util/misc/VATestprogram
- Insert VATestdisk6.25b.adf
- Use Kickstart 1.3 or 2.0, OCS or ECS

harbaum, 2024-07-19:
Sounds like the Amiga can change its fundamental video timing. I'll look at that once the Nanomig runs stable

harbaum, 2024-08-17:
I see a similar scrolling in one of the intro screens of pinball fantasies.

This means that there's a video mode being used on Amiga side that does not have the number of lines expected by the HDMI encoding of NanoMig. The image is slowly scrolling up which means that HDMI draws probably one line more than the Amiga is sending. So the Amiga for some reason decided to send less lines per screen than usually.

Currently I only differentiate between PAL and NTSC and both work fine with e.g. the Amiga test kit. So there is a video mode being used my HDMI encoder is not aware of. I wonder if it's an interlaced mode. I need to find some test program that allows me to switch all kinds of video modes. Preferably something that can auto start without interaction as I could then run this in the verilator simulation and have a closer look there.

harbaum, 2024-08-19:
It's indeed the interlacing that's causing this issue. The double scanned image has one line less in this case and the HDMI encoder needs to take that into account.

closed 2024-08-19 by harbaum

xolod79, 2024-08-21:
I have a similar problem on the demo "15 Years of Fame"  https://www.pouet.net/prod.php?which=61034

harbaum, 2024-08-21:
In order to debug it you might run it in simulation and watch the output of the frame capturing. It may report unexpected frame sizes.

retrofun, 2024-08-26:
Also affected by this: Great Giana Sisters, Jinxter

I had a look at it with GAO: the interesting thing is when switching back from an interlaced screen mode to non-interlaced.
Either there are 625 lines or 623 lines then (for PAL).

I have adapted the video_analyzer to handle this, [PR is here](https://github.com/harbaum/NanoMig/pull/20).

Successfully tested with Great Giana Sisters, Jinxter, VATestdisk6.28b, 15 Years Of Fame.

harbaum, 2024-08-26:
I actually wonder if this is a bug in the minimig. I don't think it should do this and I doubt a real amiga does this.

It may be worth trying to figure out why exactly this happens.

retrofun, 2024-08-27:
The Amiga is able to output long frames (313 lines) and short frames (312 lines), selectable in [VPOS](http://amigadev.elowar.com/read/ADCD_2.1/Hardware_Manual_guide/node0045.html).

Non-interlaced modes usually use long frames but it's also possible to select short frames.

In interlaced modes long frames and short frames are alternated (auto toggle control bit in [BPLCON0](http://amigadev.elowar.com/read/ADCD_2.1/Hardware_Manual_guide/node0022.html).

A detailed explanation can be found [here at EAB](https://eab.abime.net/showpost.php?p=1251298&postcount=13).

derschrei, 2024-09-05:
Same issue on Spaceballs' nine fingers demo (it happens with last release also).
It can be downloaded here: 

http://spaceballs.untergrund.net/files/spb-9fingersa.adf
http://spaceballs.untergrund.net/files/spb-9fingersb.adf

retrofun, 2024-09-06:
I can confirm that the screen rolls vertically for Spaceballs Nine Fingers Demo with nanomig.fs [v0.3.](https://github.com/harbaum/NanoMig/releases/tag/v0.3)
With nanomig.fs [v0.4](https://github.com/harbaum/NanoMig/releases/tag/v0.4) it works for me.

commit 8f3fbb3a (2024-08-19): HDMI adopt to scandoubled interlaced signals. Fixes #9