MiSTle-Dev/NanoMig #12: Get GAO under Linux working

harbaum, 2024-08-20:
Thanks to the great document provided by retrofan I was able to use the programmer shipped with the Linux version of the gowin IDE under Linux.

However, I am unable to connect from the gowin anylzer oscilloscope to the target under Linux:

[image]

The "Null" under Location sounds somewhat suspicious ... programming doesn't work and connecting to the target with the > button just gives some busy/progress bar thing but nothing further happens. Also clicking the "trigger now" button does not have any effect.

This looks like similar problems with the programmer software. But e.g. the logs don't give any further hints.

harbaum, 2024-08-20:
It seems to work if I run gao_analyzer as root. But I'd still like to run this as a regular user ...

retrofun, 2024-08-21:
Which version of Gowin EDA and Programmer are you using?

Programmer 1.9.10.01 seems to be buggy and selects the wrong cable, GWU2X. It should select FT2CH.

Try downgrading the Programmer. The combination Gowin EDA 1.9.10.01 & Programmer 1.9.10 works fine for me.

harbaum, 2024-08-22:
The Programmer works for me (following your instructions, thanks for that), even version 1.9.10.1. The detected cable is named "Gowin USB Cable(WINUSB)".

But the programmer inside the GAO window does not work for me. Instead I can use openFPGAloader to download the a0.fs and then the capture itself in the GAO windows does indeed work.

retrofun, 2024-08-22:
I've tried again. Yes, Programmer 1.9.10.01 now calls it "Gowin USB Cable(WINUSB)" and from the Programmer application itself this works.

GAO uses the Programmer and from within GAO the wrong cable is selected, see your screenshot. When you delete the Programmer directory, GAO fails with "Can not launch JTAG server".

As I already wrote downgrading the Programmer helps. Then it works standalone and from within GAO.
It is even sufficient to replace the _bin/programmer_cli_ executable.

retrofun, 2024-10-25:
GAO works (again) with latest Gowin V1.9.10.03 / Programmer 1.9.10.03 (Build 41017)! :+1: