MiSTle-Dev/NanoMig #17: More memory?

quartexNOR, 2024-08-23:
First, this project is amazing! You guys are so cool for making this 😎

My only concern is memory. I run a lot of new software compiled to 68k, and my devtools alone needs 64mb alone. Is there any way to get 128mb or 64mb on these devices?

Other than that -- you guys rock 🤘🙏❤️

harbaum, 2024-08-23:
These tiny and cheap FPGA boards come with 8 megabytes of memory. Expanding them is probably not worth it.

The Nanomig is not meant to be a high end Amiga. The fx68k CPU currently being implemented can even only address 16 megabytes of address space. 

It's sure possible to update it for faster speeds and more ram. But there are already platforms like mister targeting this use case.

This project aims to be as compatible to the original machines while being as cheap as possible.

But of course it's all open source and whoever thinks that the NanoMig should become a high end machine is invited to implement just that.

harbaum, 2025-01-06:
Claude has just used this to build a base for the PiSTorm (see #25). This could give you a compact yet fast and powerful amiga with plenty of RAM and a matching fast CPU.

kowoba, 2025-01-13:
Ignoring the PiStorm for a moment... 

"fast RAM" is in general way more useful than both 020 and AGA (and without any fast ram 020+AGA is severely crippled). The max 3.5 MB of ram in total in original Minimig is a constant annoyance, even just 1 MB of fast ram would make everything a lot easier. Back in the days, my A500 had 2MB of fast RAM on the side, the first 68000 Amiga with IDE interface I got my hands on, the A600, came with 4MB of fast RAM on PCMCIA.

A little Fast RAM makes it possible to _comfortably_:
- use more than just one partition on filesystem (filesystem buffers run off with RAM)
- bring the system online (PPP) and avoid the need to shuffle SD cards back and forth for transfers, wearing down both card and sockets physically
- run any version of also OS 3, including latest greatest 3.2
- use the system for creative activities such as animation and music
- use WHDLoad without running into weird corner cases because not enough continuous chip or slow ram is available

If there is 8 MB on the board, why not put all 8 MB to good use?

Mangerio, 2025-04-02:
If i understand it right, the 8 MB of the Tang Nano 20K are 32 bit ram. The A500 had only a 16 bit ram. If Adressen with 16 bit only 4 MB are accessible. 2 chip, 1,5 slow and 0,5 rom cache.
If a 68020 will be implemented it will have a 32 bit bus and all the 8 MB will be accessible. Not bad for a basic Amiga...

m-presnov, 2026-01-19:
Is it possible to add external QSPI/SPI PSRAM (4, 8 or even 16 MB)? How many LEs does it cost? Perhaps it demands 68020 with cache (burst read/write - 32 bytes minimum).

closed 2026-10-22 by Manger74