MiSTle-Dev/NanoMig #22: Latest nanomig core doesn't boot floppy images reliably anymore

fvillanustre, 2024-09-02:
There has been a regression on the reliability of the floppy image boot process with the latest release from August 31st. While I managed somehow to boot a few floppy images, there seems to need some timing sorcery to boot them. This is with the latest MOS unified firmware. After rolling back the nanomig core to the previous version, every floppy image boots reliably again. This is while using HDMI.

harbaum, 2024-09-03:
May be a build-to-build instability. Did you try building the core from scratch, again?

fvillanustre, 2024-09-03:
Good question! No, I did not. I used the build that you provided for version 0.4. I don't have the environment to build it installed (but I could certainly install it and try). If I reset it a few times, sometimes it will boot the floppy. Otherwise, the screen would stay on the standard KS 1.3 waiting for a floppy screen. But if I roll back to 0.3, every floppy image works flawlessly every time.

closed 2026-02-23 by Mangerrione