MiSTle-Dev/NanoMig #23: Hardware Component Connections

SparkyNZ, 2024-10-03:
Hi. Could you please add a diagram or photo showing which pieces of hardware connect to each other. I have a 20K, m0S and other bits and pieces here. I'd love to give this a go but other than loading firmware to the devices I wouldn't have a clue what to connect to what. I know this is early work in progress but I'd love to give it a go too :-) Thanks.

retrofun, 2024-10-04:
There's an example wiring for every option of a FPGA companion in the `README.md`:
* M0s dock/BL616: https://github.com/harbaum/FPGA-Companion/tree/main/src/bl616#example-wiring
* ESP32: https://github.com/harbaum/FPGA-Companion/tree/main/src/esp32#example-wiring
* RP2040: https://github.com/harbaum/FPGA-Companion/tree/main/src/rp2040#example-wiring

Some more information about wiring M0S dock is available in the MISTeryNano repository:
https://github.com/harbaum/MiSTeryNano/tree/main?tab=readme-ov-file#getting-started

SparkyNZ, 2024-10-05:
That's excellent! Thank you

closed 2024-10-05 by SparkyNZ