MiSTle-Dev/NanoMig #24: Can't get into OSD

SparkyNZ, 2024-11-16:
I've been using the same setup for MiSTery and the Vic 20 Nano. I can press F12 to get into the OSD for both of those but with Nanomig I only get the Kickstart 1.3 boot image (the big chunky hand holding a blue floppy).

I'm using the M0S Dock with an old 4-port USB hub.

I tried removing my USB hub, plugged the keyboard directly into the M0S Dock's USB port (via USB C to A adapter). I was able to get into the OSD and boot to Workbench.

Thinking maybe it was a power issue, I tried again with the hub and found that the Kickstart image didn't appear but I was able to get a corrupted OSD screen to appear. Instead of powering the Nano from my PC's USB port, I connected to a 2Amp USB brick but I still have the same issues.

Unfortunately I only have the 1 hub I can try at the moment but I do have another on the way that will connect directly to the M0S dock's USB-C port and I'll give that a try when it arrives in a week or two.

SparkyNZ, 2024-11-16:
I found another hub and gave that a try - same deal.

I remembered I had a USB-Bluetooth wireless adapter for the keyboard/mouse I use with my TV - tried that too - nothing.

FYI the led remains blue on the Nano - doesn't change to green. It's as if the Kickstart has frozen. If there's any other info I can provide you with to help, please let me know.

I tried a few other keyboards I have but they're all Dell USB keyboards. I have a small keyboard but no F12 key (only goes up to F10) so I couldn't try that one either.

SparkyNZ, 2024-11-17:
I'm told the Logitech K400 Plus works with Nanomig. Just ordered one.. I'll update this issue with my findings in a couple of weeks when mine arrives.

SparkyNZ, 2024-11-18:
Hi Till. Yes, correct, I haven't changed the BL616 firmware since I first loaded it. Maybe it is still power related in some way. So the OSD is a separate component that should run independent of the Core. Interesting. I wonder if there any signals I can monitor/test for life? I'm used to using SignalTap on the Intel Cyclone boards but I've only had a very brief play with the Tang boards.

If you have any signal monitoring suggestions, please let me know. It would probably be good idea for me to try building the design from the source files and see how that goes before I start fiddling. Is the OSD just like a big sprite/frame buffer that overlays/mixes with the VGA display?

I do have another 20K and M0S on the way too so before long I'll be able to have a second system to test it on.

harbaum, 2024-11-18:
The bl616/m0d does some debug output on the TX pin/pad right besides the USB connector. It should give either 921600 bit/s on later firmware or 2000000 bit/s on more recent ones.

Which firmware do you use? The fpga companion from its own repository? If not, please try the latest release there. I might actually need to update its release to support all cores incl. Stefvoss' latest.

harbaum, 2024-11-18:
... and yes, the osd is basically some kind of big sprite. It syncs to the amigas screen. So as long as the Amiga side is able to display anything, even broken stuff, the osd on top should be visible.

It somewhat sounds to me you are using a bl616 firmware that does not include nanomig support. This is either from a source that never included my code or it's a very old version of mine.

SparkyNZ, 2024-11-18:
The FPGA Companion was the latest at the time from here:
`url = https://github.com/harbaum/FPGA-Companion.git`

According to the git log, here's what I have on my machine:
```
E:\NanoMigHome\Documents\FPGA-Companion>git log
commit 762e0abd1cf626e8ea88145449f67b0021b06ad7 (HEAD -> main, origin/main, origin/HEAD)
Author: Till Harbaum <till@harbaum.org>
Date:   Tue Oct 29 08:48:00 2024 +0100

    Add entry to enable Amiga IDE
```
Your Amiga OSD code is definitely included because the OSD does work when I remove the USB hub.

I'm about to start work so I don't have time to try getting debug from the M0S at the moment, sorry - but I can give it a try this evening. 

Are those pads just TTL/RS232? I'm assuming I can just connect an RS232-USB converter on those 3 pins and get some serial debug out with PuTTY?

harbaum, 2024-11-19:
The Amiga OSD _does_ work when no USB hub is involved? But with the very same hub the other cores do work? Odd ...

And yes, it's just ttl uart at 3.3v.

SparkyNZ, 2024-11-20:
Correct, the OSD does work when I plug the keyboard straight into the M0S. And yes, the same hub works fine with the other cores.

Anyway.. something incredible has happened - I soldered a 3 pin header onto the Tx/Rx/Gnd pads of the M0S dock, connected PuTTY and the TTL-to-USB board.. and now the mouse and keyboard are working through the hub????

I disconnected the TTL-to-USB and now it isn't working..
I reconnected the TTL-to-USB..  no longer working once again

So I had a few minutes of progress followed by a complete regression. Before it stopped working, the OSD was flickering and corrupting like mad. I have 25Meg video.. I'll see if I can attach it somehow.
[image]
[image]

If I press F12 to get the OSD, it appears on screen, corrupted.
[putty.log](https://github.com/user-attachments/files/17823720/putty.log)

closed 2024-11-22 by SparkyNZ

harbaum, 2024-11-22:
To be honest ... That makes no sense to me :-) But if it works ... Who cares?

Anyway, the TP25k should actually be easy to support. I already support that with the MiSTeryNano.

Did you also buy the sdram board? The TP25k itself has no ram at all. Unfortunately that sdram eats a lot of Io leaving only the three ports which are needed for HDMI, SD card and the bl616/whatever which in turn leaves nothing for e.g. midi or classic digital joysticks.

harbaum, 2024-11-22:
Regarding aga: I haven't yet managed to run the tg68k any faster than 8 MHz which is a prerequisite for aga unfortunately ...

SparkyNZ, 2024-11-22:
> To be honest ... That makes no sense to me :-) But if it works ... Who cares?

I don't care anymore :-) I'll be recommending the cheap 3 port hubs from Aliexpress. Shame I bought one of those Logitech keyboards too.. I'll have to sell that when it arrives.

> Anyway, the TP25k should actually be easy to support. I already support that with the MiSTeryNano.
>
> Did you also buy the sdram board? The TP25k itself has no ram at all. Unfortunately that sdram [...]

Yeah I have the SDRAM board. That's usually the problem with FPGA boards - pins used up by other crap. Unfortunately all of my other FPGA dev boards only have Cyclone IV which have about 10LEs max. I do have a Cyclone V but that's for my own tinkerings. :-)

I guess the advantage of the original Minimig having its own CPU is that is frees up a bunch of LEs? That doesn't help with pins though. I knocked up an SDRAM controller a couple of years ago and yeah - that's a lot of pins!