MiSTle-Dev/NanoMig #25: Not directly a issue :)

captain-amygdala, 2025-01-05:
Just got your awesome Nanomig on Tang Nano 20k working with a Pistorm interface.
 I thought you might find it funny/interesting :)
 Greetings, Claude (The Pistorm guy) 

[image]
[image]
[image]

harbaum, 2025-01-06:
Hey Claude! Welcome! And yes, really awesome. 

This allows to expose the original Amiga ports on physical pins. Adding joystick and mouse ports is probably the most forward usage. But one could e.g. try to add the printer port or even connect a real floppy drive.

captain-amygdala, 2025-01-06:
Hi Till,
yes I plan to make a small PCB for it.
I actually did already one , but forgot to connect some essential GPIOs from the Pi, so a new revision is needed :)

I don't quite understand the physical pins for joystick,mouse and printerport , because unfortunatley the Pistorm interface to the Pi needs 24 GPIOs from the Tang Nano. So both the Pi as the Nano have almost no free GPIOs. I also had to remove the SD card support, as some pins are shared between the SD card and the Tang GPIOs :( When I do a new PCB, i might add already a RP2040/Pico to the PCB, would it be feasible to shift the entire SD handling, and maybe also kickstart loading to the RP2040 ? Adding a SD slot to the RP2040 is not much work.

harbaum, 2025-01-07:
AGA will probably fit without the CPU. The main hurdle so far was that I didn't yet manage to run the tg68k CPU in 020 mode at twice the speed as a faster CPU is needed for most AGA software. But with the PiStorm this is not an issue. So yes, AGA may be possible. This requires a faster memory bus interface which actually shouldn't be too difficult to implement as the TN20k's ram is already 32 bits wide (only 16 used by now) and it should also be possible to enable burst access with the ram.

captain-amygdala, 2025-01-13:
Designed now a new PCB. RP2040 and USB hub integrated. The Board is ~15€ assembled and shipped to EU from JLC.

[image]

JJDSNT, 2025-01-13:
That is awesome! thanks people! kudos!

Maverick-Shark, 2025-01-16:
> Designed now a new PCB. RP2040 and USB hub integrated. The Board is ~15€ assembled and shipped to EU from JLC.
> 
> ![capture-2025-01-13T13_07_46 024Z](https://private-user-images.githubusercontent.com/16537586/402555207-db03039f-e707-45f2-8c0e-035500c8c703.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzcwMjE1NTEsIm5iZiI6MTczNzAyMTI1MSwicGF0aCI6Ii8xNjUzNzU4Ni80MDI1NTUyMDctZGIwMzAzOWYtZTcwNy00NWYyLThjMGUtMDM1NTAwYzhjNzAzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE2VDA5NTQxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxMzY3ZmY2YzQwZWE3MjY4Y2Q0NzE1YzBlZTlkMTU3ZGIzOTE5NTE5N2Y2MGEwOGRmM2IxNjRjNDEwYjhkMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0tUxkjegSRRD6WaS1rf4EGltP5I6dtaL_sOpfjljF60)

Awesome work!!

Where could we find the gerbers or kicad files?

Thanks,
Shark

harbaum, 2025-01-16:
Indeed, great work. If only JLCPCB would be able to provide the FPGA as well for reasonable prices. I've tried to get in touch with Gowin regarding this, but no answer so far.

Interestingly Efinix does not have those nice Tang Nano like boards (especially with SDRAM/PSRAM, SD card and HDMI). But their chips are way easier available _and_ they do communicate. I am thus trying to get the NanoMig running on a Efinix Trion T20. Biggest hurdle is HDMI and this is what i am thus trying first. Unfortunately I haven't had any success, yet. The differential LVDS serializer cannot be used to drive 3.3V signals generating a TDMS signal "manually" in pure verilog looks better but is not accepted by my screens, yet. 

The only example for HDMI/DVI on a Efinix I've found so far uses a LVDS-to-TDMS bridge chip which I'd like to avoid.

JJDSNT, 2025-01-17:
hi, dunno if i understood right, but If it doesnt have hdmi why do you need hdmi (as in the pistorm land it uses amiga rgb)?

captain-amygdala, 2025-01-17:
> Interestingly Efinix does not have those nice Tang Nano like boards (especially with SDRAM/PSRAM, SD card and HDMI). But their chips are way easier available _and_ they do communicate. I am thus [...]

Hah you can read my mind ? :) @harbaum 

I'm sitting currently in front of my Pistorm32 design , which uses a Trion T8Q144/T20Q144  and wondering on which bank I should add the SDRAM to the Trion . I go probably a different route in this case regarding video. I have written a MIPI CSI2 TX Core for the Tang9k which uses a passive resistor network on normal IOs, this I plan to port over to the Trion. With MIPI I can directly DMA into the raspberry hdmi buffer trough the Pi camera port, so the Minimig Video can be outputted to the Pi HDMI with near zero latency and use the Pi hardware scaler.

The good thing on the Trions is that JLCPCB can order these for cheap and assemble them.
Total BOM cost would probably even be lower than what a Tang20k cost. 

Regarding HDMI on Trion, instead of LVDS Serializer, how about a DDR IO ? That's what I use(ed) for MIPI as well. Serialisation factor is just 2 then , but half the clock rate into comfortable for FPGAs regions

captain-amygdala, 2025-01-17:
Just to add. I designed a prototype SDRAM + RP2040 + HDMI expansion board for the Pistorm32lite a while ago with minimig in mind.  But I somehow don't like this "Sandwich" design with multiple boards. Therefore I started some days ago a more compact version on a single board with a CM4 Raspberry module . 

[image]

harbaum, 2025-01-17:
Yes, the double data driver is an option in order to not need twice the HDMI clock.

However, a barely filled T20 should IMHO be able to manually produce a usable result with twice the HDMI clock for scan doubled Amiga. So I tried to generate all eight TMDS signals "manually" at 282Mhz. With my limited tools at home I see something that actually looks pretty similar to the same signals on the Tang nano. However my screen does not like it. This may be a jitter problem or polarity or bit order or a combination of this ... this is tricky to debug. I already considered using a second FPGA to capture this signal to be able to debug it ...

captain-amygdala, 2025-01-17:
If you want, I can try helping to debug it. Have a scope here which should be able to single-shot a 283MHz signal, also some experience with hdmi/dvi on FPGAs.

harbaum, 2025-01-17:
This all sounds like we should cooperate in general :-)

You know my personal email? Please drop me a message ...

harbaum, 2025-01-17:
> hi, dunno if i understood right, but If it doesnt have hdmi why do you need hdmi (as in the pistorm land it uses amiga rgb)? 
> (HDaMI video pass-through)

Because this.also about the regular NanoMig which obviously needs HDMI. And having the option to run this pi daughter board stand-alone without Pi might also be nice.