MiSTle-Dev/NanoMig #26: Same pin layout for DB9 mouse/digital joystick as the MiSTeryNano?

kragil, 2025-01-05:
Hello, I saw your talk at C3. Great stuff!

Can I use the same pins as seen here:
https://github.com/harbaum/MiSTeryNano/raw/main/images/wiring_spi_midi.png

for DB9 on NanoMig?

So I could just connect wires from my breadboard to my old Amiga Joystick in a really janky way and it should work?

harbaum, 2025-01-06:
Thanks, you are welcome!

And yes, the joystick should just work. If you don't need 5 volts for your joystick (e.g no magic auto fire, just a bunch of passive switches) then you can wire it directly to the FPGA and it will work. If you need 5V then you might have to use level shifters to protect the FPGAs IO lines from the 5V the joystick may return.

This can even be rewired inside the core to the other port to use a real amiga mouse.

dijitalxyz, 2025-01-07:
Are there additional connections for a physical DB9 joystick?
I think it will support the same structure as the Wiring at https://github.com/harbaum/MiSTeryNano/

harbaum, 2025-01-07:
For a second joystick you'll have to change the pin usage on the FPGA to get additional pins free.

closed 2026-10-22 by Manger74