MiSTle-Dev/NanoMig #28: kick31.rom stuck

dijitalxyz, 2025-01-07:
kick13.rom F6290043
openFPGALoader --external-flash -o 0x400000 kick13.rom
openFPGALoader --external-flash -o 0x440000 kick13.rom
1.3 hand on boot work 

kick31.rom 1483A091
openFPGALoader --external-flash -o 0x400000 kick31.rom
black screen stuck on boot

harbaum, 2025-01-07:
With the latest code from Git or with the very old release?

dijitalxyz, 2025-01-07:
not recompile latest GIT
I use release nanomig.fs  Sep 1, 2024

harbaum, 2025-01-07:
I just did a new release which for me boots kick 3.1 as expected.

This release also works with rdb Harddisk images.

dijitalxyz, 2025-01-08:
I found the faulty part
Kickstart3.1.rom FC24AE0D is ok

closed 2025-01-08 by dijitalxyz

harbaum, 2025-01-08:
For the HDF image you might try following this guide:

https://namelessalgorithm.com/amiga/blog/genhdf/

harbaum, 2025-01-08:
And here is an empty HDF generated with `xdftool` and `rdbtool`:

[empty.hdf.gz](https://github.com/user-attachments/files/18349168/empty.hdf.gz)

As it's empty you won't be able to boot from it. But you should be able to install workbench from floppy disk onto it.