MiSTle-Dev/NanoMig #29: may be a problem with HDMI in the newly compiled firmware

dijitalxyz, 2025-01-09:
There may be a problem with HDMI in the newly compiled firmware
There is no problem with LG TV or monitor TV
but the same setup

release 0.4 + BIOS 1.3 + fpga_companion_pico_and_pico-w.uf2 1.0 version
Capture image from HDMI capture card OK

release 0.5 + BIOS 1.3 or 3.1 + fpga_companion_pico_and_pico-w.uf2 1.1 version
Capture image from HDMI capture card FRAME DROP
I tried 3 times by re-flashing with the same FPGA card in the same time period on the same system
it seems that the only change is the new fpga firmware

harbaum, 2025-01-09:
Hmm ... I don't think I've changed the video timing in any way. Can you try to figure out which change during 0.4 and 0.5 affects this?

harbaum, 2025-01-09:
There were changes that were related to interlace modes:

https://github.com/harbaum/NanoMig/commit/6bc8bd6c27f7edd5310715f4ed6f5286315d57f4

You don't use interlace modes, I'd assume. Anyway, undoing these step by step could help finding the issue

mibosshard, 2025-01-09:
PAL/NTSC? how many frames dropped? constantly or every now and then?

I tried it with a Blackmagic UltraStudio Recorder 3G: no chance. the damn thing won't sync at all.
the Elgato Cam Link 4K on the other hand seems to work just fine. got a recording with 768x576@49.917fps. no time right now to investigate further. played back the recording with LibreELEC on a SAMSUNG public display. it looked as if playback stutters most minimal every now and then. I have now idea if it's the display or the recording... I have an ASUS display for all my retro needs which syncs to almost any weird frame rate. but as I said: no more time right now...

dijitalxyz, 2025-07-03:
4l2-ctl -d /dev/v4l-subdev2 --query-dv-timings
	Active width: 768
	Active height: 576
	Total width: 908
	Total height: 626
	Frame format: progressive
	Polarities: -vsync -hsync
	Pixelclock: 28420400 Hz (50.00 frames per second)
	Horizontal frontporch: 0
	Horizontal sync: 140
	Horizontal backporch: 0
	Vertical frontporch: 0
	Vertical sync: 50
	Vertical backporch: 0
	Standards: 
	Flags: