MiSTle-Dev/NanoMig #31: Nanomig V0.5 virtual hd

fhersete, 2025-01-14:
I have added the 3.1 rom and it works correctly, the df0 and df1 disks too, but when I try to load a virtual hard disk in hdf format from os1.3, 2.0 or another, none of them start. What are the loading instructions for a virtual HDD? I have tried with DH0.hdf in root, but there is no way for it to start, I have even disconnected df0 and it doesn't.

Greetings.

mibosshard, 2025-01-15:
in the settings menu (IDE) "Enabled" and "Disabled" are mixed up. meaning if IDE is set to "Enabled" it is in fact disabled and vice versa.

harbaum, 2025-01-15:
I'll check that enabled/disabled thing which is quite surprising as I am sure I use that setting every now and then and haven't noticed that, yet.

dijitalxyz, 2025-01-15:
(IDE) If we assume that we have solved the "Enabled" and "Disabled" problem
What device driver is recommended to be installed or what configuration is used for the HD0 image file?
SCSI_DEVICE_NAME=scsi.device
SCSI_DEVICE_NAME=uaehf.device

mibosshard, 2025-01-15:
> I'll check that enabled/disabled thing which is quite surprising as I am sure I use that setting every now and then and haven't noticed that, yet.

I just double-checked this (Tang Nano 20K + M0S) with latest firmwares (NanoMig v0.5 + FPGA Companion v1.1).

deleted amiga.ini on the µSD card: system doesn't start and sits on a white screen, unable to invoke OSD.
created an empty "amiga.ini" file on the µSD card: all good, system starts to Kickstart screen.
per default IDE was set to "Disabled". configured "Harddisk DH0:" and did a reset (OSD): system boots from HDF.
then changed the IDE setting to "Enabled" and did a "Save settings, powered off the Tang Nano. plugged in power and the system starts only to the Kickstart screen.

I checked the relevant amiga.ini variable:

var I=1 (HDF boot OK)
var I=0 (no boot from HDF)

mibosshard, 2025-01-15:
> (IDE) If we assume that we have solved the "Enabled" and "Disabled" problem What device driver is recommended to be installed or what configuration is used for the HD0 image file? SCSI_DEVICE_NAME=scsi.device SCSI_DEVICE_NAME=uaehf.device and image files are always in readonly mode (no write support) df0 , df? hd0 or hd? etc.

Enabled/Disabled IDE: not a problem per se as soon as one knows about this tiny mixup...

since it's working with a patched (added scsi.device) Kickstart1.3 I'd assume scsi.device. Kickstart 2+ should work anyway.

AFAICT: ADFs are RO and HDFs are RW.

mibosshard, 2025-01-21:
here's an empty (but bootable) image (HDF - 2.0 GB): 

[2GB_FFS_EMPTY_BOOT.HDF.zip](https://github.com/user-attachments/files/18495308/2GB_FFS_EMPTY_BOOT.HDF.zip)

dijitalxyz, 2025-01-22:
minimig:
it doesn't start when I open it with the bootable hdf file you gave

The only change from your system is that I use Pi Pico.

on emulator
winaue:
copy of this file at boot

when I want to see it, it doesn't see a copy of this file as a hard disk 
but when I boot the system with adf, it comes as hdf and w.1 is installed

dijitalxyz, 2025-01-23:
I couldn't boot with any HDF file that I had or that came from outside
Even when I booted with ADF, there was no indication that the HD0.HDF file was attached to the system
can someone verify this for me?
I know these checksums are correct but I don't know if I started with the right file

My installed working versions

kick3.1 = 40.63
NanoMig 0.5 TANGNANO 20K lastest
FPGA-Companion 1.1  fpga_companion_pico_and_pico-w.uf2  lastest 

Kickstart 3.1 version 40.63 
FC24AE0D	e40a5dfb3d017ba8779faba30cbd1c8e *kick40063.A600

               Kickstart
 Amiga Model   Version              P/N(s)
 --------------------------------------------------------
 CD32          (40.56/40.58/40.60)  391640-02/?/391640-03
 500/600/2000  (40.63)              ?
 A1200         (40.68)              391773-01, 391774-01
 A3000/3000T   (40.68)              ?
 A4000         (40.68)              ?
 A4000T        (40.70)              391657-01, 391658-01
 CD1200        (40.70)              ?

may be I need scsi.device 40.21 for A600
Kickstart update (40.72) from Hyperion Entertainment:

dijitalxyz, 2025-04-30:
one last question @mibosshard @harbaum 
can you share the checksum information of the version you used as bios 3.1?
i gave up because i couldn't get through with the system i had in the last try
maybe i can achieve success with the new version

mibosshard, 2025-05-01:
as far as I can remeber, I never used Kickstart 3.1 on the NanoMig. the Kickstart 3.1 I use on the MiSTer: MD5 0b839c665635a249c5736118c68a69a7

dijitalxyz, 2025-05-02:
Thanks for the quick reply.
So let me examine what's available a little bit.

dc10d7bdd1b6f450773dfb558477c230 *Kickstart v2.04 rev 37.175 (1991)(Commodore)(A500+).rom
c3bdb240 *Kickstart v2.04 rev 37.175 (1991)(Commodore)(A500+).rom

5f8924d013dd57a89cf349f4cdedc6b1 *Kickstart v3.1 rev 40.60 (1993)(Commodore)(CD32).rom
1e62d4a5 *Kickstart v3.1 rev 40.60 (1993)(Commodore)(CD32).rom

0b839c665635a249c5736118c68a69a7 *Kickstart v3.1 rev 40.63 (1993)(Commodore)(A500-A600-A2000).rom
88136ca9 *Kickstart v3.1 rev 40.63 (1993)(Commodore)(A500-A600-A2000).rom

646773759326fbac3b2311fd8c8793ee *Kickstart v3.1 rev 40.68 (1993)(Commodore)(A1200).rom
1483a091 *Kickstart v3.1 rev 40.68 (1993)(Commodore)(A1200).rom

9bdedde6a4f33555b4a270c8ca53297d *Kickstart v3.1 rev 40.68 (1993)(Commodore)(A4000).rom
d6bae334 *Kickstart v3.1 rev 40.68 (1993)(Commodore)(A4000).rom

dijitalxyz, 2025-05-17:
Has anyone tried using DH0: DH1: at the same time?