MiSTle-Dev/NanoMig #34: Networking via MCU?

Mangerio, 2025-03-31:
The BL616 and the ESP32 MCU's offer WiFi capability.
Is or will this be accessible from the AmigaOS/FPGA side?

harbaum, 2025-04-01:
This has actually been implemented for the Pico-W and the ESP32 and is being implemented for the Atari ST core.

This actually doesn't implement a network device for the core but a serial device reachable via the targets rs232 port. The Atari ST can use it to dial into BBS's. Something similar has been done for the C64.

Since I am no Amiga user and never owned one I wouldn't even know how networking and bbs'ing worked there back in the days.

The bl616 should support WLAN as well. But the SDK was too broken to support wifi in a usable way last time I tested this a few months ago.

Mangerio, 2025-04-01:
Thank yor for the reply!
Maybe networking is not so important for the NanoMig right now.
Basically it's an A500 with some extra RAM. 
It's fun to play games and watch demos!
In the current state it would be underpowered for a full network stack i guess...

closed 2025-04-06 by Mangerio