MiSTle-Dev/NanoMig #36: DB9 Joystick Issue

Mangerio, 2025-04-03:
Just wired a DB9 socket to the NanoMig.
Have wired Up, Down, Left, Right, Button 0 and Ground.
No need for button 1 as my Competition Pro has both buttons on 0.
Tested everything carefully...

Now i am experiencing a strange behavior.
Up, Down, Left, Right give no reaction in Joytest & Games.
Button 0 is recognized.

The strange thing is, when i hold the button down,
Up, Down, Left and Right are working!!!

The Joystick has no electronics. Just switches.
Same behaviour with an other Competition Pro model.

What can be wrong?

ps
NanoMig is soooo much fun!
The chaos (cable salad) on the table! 
Love it! :- )))) 
Thank you!!!

harbaum, 2025-04-03:
Sounds like you got the cabling wrong. Please check if you have e.g. GND and the Button0 pin swapped.

Mangerio, 2025-04-03:
You were right!
Stupid mistake, i'm sorry!
Now the games can begin!!!

By the way... is there a reason, why you did not implement the other joystick port?
Many games are made for two players. Like one of my favourites... IK+
Is it difficult to add another one?

closed 2025-04-04 by Mangerio