MiSTle-Dev/NanoMig #41: NanoMig RAM

Mangerio, 2025-04-23:
Yesterday i got AGS ECS 2.7a (Amiga Game Selector) running on the NanoMig. It's a collection of 14 gigabytes of games, demos and apps in one HDF file. The ECS version was optimized for the A500/2000 and A600. So it would be perfect for the NanoMig. It runs fine, except for the RAM usage. It expects some megabytes of fastram...

No complaints!
I just would like to understand how the RAM of the NanoMig works...

The Tang Nano 20K offers 8MB of 32bit RAM.
If i understand it right, a 16 bit system (68000+ECS) can use only half of it? For example 2MB chip + 1,5MB slow + 0.5MB for the ROM?

Could an 68020+AGA system use all of the 8MB?
Like 2MB chip + 5MB fast + 1MB ROM?

That would be sufficient for most Amiga distributions...

harbaum, 2025-04-26:
Actually, the core could access all 32 bit RAM. The Atari ST core does that giving the Atari a total of 8MB.

In Amiga world, there's IMHO only up to 3.5MB "regular" ram. Everything else needs to use this autoconf thing. And since I am no Amiga expert at all, I simply didn't add this as I have no idea how that works.

The NanoMIG core definitely needs some support by an FPGA experienced Amiga fan. 

And yes, the AGA 68020 variant may include that as well. But I am still not sure it's even possible to fit that into the Tang Nano 20k.

Mangerio, 2025-04-27:
That's good news! An A500 with 2MB chip and 6MB fast wouldn't be bad at all! 
I believe an AGA-020 core exists for the MiST? But the MiST has some thousand LE more than the Tang Nano 20K.
If AGA doesn't fit it would be nice to clock the 68000 to 40 MHz as an option...

The NanoMig is really much fun!
It would be nice if someone could take care of it!
If it is okay for you, i will try to find someone who can do aufoconfig RAM. Maybe someone is just waiting for a challenge like this?

Mangerio, 2025-05-23:
I just had a conversation with Alastair M. Robinson, a well known Amiga and FPGA expert, about autoconfig Ram.

"Autoconfig's not difficult - the MiSTer core has it integrated into the CPU Wrapper - but the MiST / TC64 core (which I work on) has an Autoconfig module as part of the chipset, which should be very easy to integrate and modify to add the appropriate RAM size. Then it's just a matter of mapping the correct memory range to the right part of SDRAM."

He might also code a Fastram Module for the NanoMig, if he can find time for it.

"Yes, I'll see what I can do - but I can't make any promises timewise."

Mangerio, 2025-05-24:
The NanoMig is using the MiSTer MiniMig AGA codebasis.
If i understand it right, in the "cpu_wrapper.v", a fast RAM memory extension is present and needs to be configured properly?