MiSTle-Dev/NanoMig #45: More details on required kickstart versions

danboid, 2025-05-17:
The readme states kickstart 1.3 and optionally kickstart 3.1 are required to use nanomig. Should absolutely any version of these kickstart versions work? I have have seen several variations of both kickstarts online and my first attempt at flashing both has failed.

If specific versions are required then could you please provide checksums and be more specific in the README.

I can use kickstart 3.1 on my nano 20K to get IDE support working right?

Is it simlating an A600? I presume it is simulating a specific model. Please add this to the README too.

danboid, 2025-05-17:
I can partly answer my question now I've tried it on another display.nanomig simulates an A500, an A1000 and "ECS" which I presume is equivalent to an A600?

danboid, 2025-05-17:
I realise the AROS kickstarts have compat issues with games but are the AROS kickstarts known not to work?

danboid, 2025-05-18:
I've got a working 1.3 KS.

sha265sum:

ee05862d8102a08436ac4056da7d549db31625c7d47b24dfb7b3c9a5c113ca53  Kickstart-v1.3-rev34.5-1987-Commodore-A500-A1000-A2000-CDTV.rom

danboid, 2025-05-18:
I've not been able to get KS 3.1 to work yet. I presume the version we should be using is

Kickstart v3.1 rev 40.63 (1993)(Commodore)(A500-A600-A2000).rom

f94228579fba5d26146a9c32635edb04584f5b24eb57c87756c966694d006621  Kickstart v3.1 rev 40.63 (1993)(Commodore)(A500-A600-A2000).rom

dijitalxyz, 2025-05-18:
https://github.com/harbaum/NanoMig/issues/31#issuecomment-2846989467
may be check this

danboid, 2025-05-18:
Thanks @dijitalxyz !

Yes,

e40a5dfb3d017ba8779faba30cbd1c8e  Kickstart v3.1 rev 40.63 (1993)(Commodore)(A500-A600-A2000)[!].rom

Works for me.

danboid, 2025-05-18:
I'll wait for @harbaum to respond but I'm happy to submit a PR to get this info added to the README.

danboid, 2025-05-18:
This is getting OT slightly but I noticed that the Nano 20K has an open source hardware logo on the box.

Are the gerber files for its PCB available for download? Maybe it just has that logo on the box because its using a RISCV CPU?

harbaum, 2025-05-19:
I haven't been using any specific ROM. I just grabbed the first one that seems useful. It identifies itself as 3.1 ROM 40.063 and has the md5sum of e40a5dfb3d017ba8779faba30cbd1c8e

Maybe you also flashed that twice? Those bigger kickstarts only have to be flashed once to 0x400000

harbaum, 2025-05-19:
It's implementing an a500 with the IDE being compatibly to the one of the A600 as far as I know. I am no Amiga user at all and cannot tell much which original system had what peripherals and was using which ROMs.

fboldog, 2025-05-19:
I suggest use a correct rom for the selected chipset. 
So for OCS and ECS always use the A500 / A600 / A2000 rom. 
If you have ordered an original OS3.2 the rom name is `kickCDTVa1000a500a2000a600.rom` and with the latest update (3.2.3) the name will be `CDTVA500A600A2000.47.115.rom`

danboid, 2025-05-20:
I tried running sysinfo under nanomig running in ECS mode today but sysinfo doesn't try to guess the Amiga model its running on.

Going off what @harbaum says, the ECS chipset mode should be closer to an A500+ than an A600.

There is an Amiga library called identify that should tell us what model it thinks the ECS chipset mode represents:

https://codeberg.org/shred/identify

Unfortunately I've not found a .adf disk image for it and so I might have to install Workbench to run it. My extreme laziness resulted in me emailing its author to ask if he knows of an .adf version of identify or maybe he knows of a better program to identify Amiga models?

We could just add to the README

"Nanomig supports A500, A1000 and ECS chipsets"

if the ECS mode is neither an A500+ nor an A600, which seems to be the case.

dijitalxyz, 2025-07-05:
I saw that the ECS game selector did not work, but detailed testing is necessary
I do not think our system is really fully ECS compatible

fhersete, 2025-07-30:
> > Vi que el selector de juegos de ECS no funcionaba, pero es necesario realizar pruebas detalladas. No creo que nuestro sistema sea realmente totalmente compatible con ECS.
> 
> Ya tengo AGS 2.7a funcionando en la NanoMig con una tarjeta formateada ExFat Blocksize de 512 kb. La NanoMig es un poco exigente con los archivos HDF. Creé un HDF vacío con HST-Imager. Luego, copié todos los archivos de la imagen AGS original con WinUAE. Casi todo funciona, excepto los juegos que requieren cierta cantidad de FastRAM.

Can you upload the functional HDF?

Mangerrione, 2025-07-30:
There is some issues with the ExFat filesystem unsolved...
https://github.com/harbaum/FPGA-Companion/issues/77

But you can download the image and try if you like...
https://www.atari-forum.com/viewtopic.php?p=483022#p483022
scroll down a little...

danboid, 2025-08-22:
> [@danboid](https://github.com/danboid) and [@dijitalxyz](https://github.com/dijitalxyz) I’ve started https://github.com/MiSTle-Dev/NanoMig/blob/main/doc/KICKSTART_ROMS.md based on the information in this thread. Feel free to polish or add details. 👍

Thanks @darius600 !

I'll submit a PR to add a link to that and I think that'll do for me.

danboid, 2025-08-22:
I neglected to add a newline in front of my link to the kickstarts file but someone else can add that next to they edit the README if it is bugging them. I don't want to make another PR just to add a newline.

harbaum, 2025-08-22:
Are you interested in doing more documentation in the wiki? You can get direct write access if you wish.