MiSTle-Dev/NanoMig #46: Analog Stereo Audio Out (Suggestion)

Mangerrione, 2025-06-07:
The NanoMig core / Amiga OS are not designed for Midi.
When we use the NanoMig with tandard computer monitors on the DVI port, it would make sense to use the Midi pins for analog stereo audio out instead. Could be a nice feature!

Mangerrione, 2025-06-09:
AFAIK the LCD variant of the NanoMig outputs the mono analog audio from the Tang Nano 20K. I understand that the IO on the board is limited and that different cores have different IO requirements. Would it hurt if the NanoMig core outputs audio signals on the same ports? I'd wish a variant of the NanoMig that supports a 2nd DB9 port in the spare pins and analog audio out in the Midi pins eventually....

luciust, 2025-06-09:
> The NanoMig / Amiga OS are not designed for Midi. When we use the NanoMig with standard computer monitors (DVI), it would make sense to use the Midi pins for analog stereo audio out instead. Could be a nice feature!

I'm using DVI monitor (the same 15kHz NEC 1970NXp that uses VGA from Amiga/ST) and along - a HDMI audio extractor (~$10) - that gives you two things:
- allows for HDMI to DVI, since Tang Nano doesn't support HDMI->DVI at this time
- allows to extract stereo audio - analog. I feed it to monitor's speakers

Hope this solves you issue

closed 2026-10-22 by Manger74