MiSTle-Dev/NanoMig #49: GOWIN FPGA Designer Build Error

Mangerrione, 2025-08-04:
I've just tried to build a binary file for the TN20K in the GOWIN FPGA Designer on Windows 10.
Place & Route gets stuck in an error: **"ERROR (PR0004) : There are total 0 unrouted nets"** 
I could successfully build the "nanomig_tp25k" but not the "nanomig"...
What could be wrong?

[image]

harbaum, 2025-08-04:
Haha, very funny. Same here with 1.9.11.03

I suggest using an older version, as that doesn't make much sense to me.

darius600, 2025-08-04:
No issues here on Win 11 with V1.9.11.03_Education_x64

darius600, 2025-08-05:
I have to correct myself.

I get “**ERROR (PR0004) : There are total 0 unrouted nets**” for all of the following:
- V1.9.11.03_Education_x64 (Build 81398) on Win11
- V1.9.11.03_Education_x64 (Build 81398) on Ubuntu-24.04 in Win11 WSL
- V1.9.11.03_x64 (Build 81277) on Win11 

For my above reply I had used a modified source base with minor changes.

Looking at these changes and the TA1132 warning messages I found that the error disappears when removing line 167 in top.sv (`.spi_io_clk(m0s[3]),`).

harbaum, 2025-08-16:
I've reported this to Gowin ... we'll see

darius600, 2025-08-19:
With the latest release [v0.7](https://github.com/MiSTle-Dev/NanoMig/tree/v0.7) building with Gowin_V1.9.11.03 Education worked for me.