MiSTle-Dev/NanoMig #54: Problems running Verilator

Mangerrione, 2025-08-20:
Have installed Ubuntu in a VirutalBox VM to run Verilator...
SDL seems not to be setup correctly...
Got exactly the same error under WSL2...

What could be wrong?

have SDL installed like this: sudo apt-get install libsdl2-2.0-0

Errors:

/bin/sh: 1: sdl2-config: not found
/bin/sh: 1: sdl2-config: not found

and

../nanomig_tb.cpp:17:10: fatal error: SDL.h: No such file or directory
17 | #include <SDL.h>
| ^~~~~~~
compilation terminated.
make[1]: *** [Vnanomig_tb.mk:67: nanomig_tb.o] Error 1
make[1]: *** Waiting for unfinished jobs....
rm Vnanomig_tb__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/mnt/c/Users/Gerald/NanoMig/sim/obj_dir'
make: *** [Makefile:54: nanomig] Error 2
root@Computer:/mnt/c/Users/Gerald/NanoMig/sim#

Mangerrione, 2025-08-20:
Almost there...
but it seems to be an error in the fx68kAlu.sv line 313 ???

[0] %Error: fx68kAlu.sv:313: Assertion failed in TOP.nanomig_tb.nanomig.cpu_wrapper.cpu_inst_o.excUnit.alu: unique case, but none matched for '32'h00000000'
%Error: fx68x_verilator/fx68kAlu.sv:313: Verilog $stop
Aborting...
make: *** [Makefile:61: run] Error 1

Mangerrione, 2025-08-20:
Thanks! Have tested the "fx68x_verilator" from NanoMac and from MisteryNano, but still get the same error....

darius600, 2025-08-20:
Which Verilator version are you using?
I just tested with Verilator 5.030 2024-10-27 rev v5.030 (build from source under WSL2 Ubuntu) and compiling & running the testbench works after fixing:
`
%Error-PINNOTFOUND: nanomig_tb.v:123:5: Pin not found: 'joystick' : ... Suggested alternative: 'joystick0' 123 | .joystick(6'b000000), | ^~~~~~~~`

[image]

Mangerrione, 2025-08-21:
Thank you for the hint, 5.03 works!
I was using the latest verilator 5.38 before...

But it is very slow...
from power on to the Kickstart Screen 3 and a half minutes....
is this what we can expext from simulation?

harbaum, 2025-08-21:
That's actually pretty fast for a full hardware simulation. It simulates all the connections of the real hardware and can write every state and state change of every single signal into trace files.

What did you were hoping this is for?

The simulation is not meant to be some kind of emulator or so. It's mainly useful for debugging. On real hardware there's no way to observe the entire device. In a basic setup you basically don't have any access to internal signals. Things like signal tap or gao give you access to a few signals over a very limited period of time. The simulation instead gives you access to all signals to any time you are willing to wait. 

Do you want to know how many clock cycles it takes between the Amiga os accessing the Paula registers and the actual floppy data being read? Do you want to trace the generation of audio data? Do you want to understand when exactly the  rom overlay is disabled by Amiga os? These are the things the simulation allows you to see in all detail, debug and fix.

Mangerrione, 2025-08-21:
I'm new to FPGA development, so i did not know what i could expect from the simulation.
At the moment i try to learn how the menu system works.
My idea would be a menu item for swaping the DB9 ports.
I thought i could test it in simulation, but the MCU isn't simulated anyways...
Now i know! :- )
Sim is about analyzing signals! 
Understood!

closed 2025-08-21 by Mangerrione

darius600, 2025-08-22:
Found it.

Starting with Verilator 5.038 the default behavior for handling `unique` statements changed: https://github.com/verilator/verilator-announce/issues/77
Earlier version did run without these assertions per default. I also tested with 5.036 which does not show this issue.

Workarounds are to **use Verilator up to 5.036** or to **add the parameter --no-assert** to the Makefile in line 53:
`verilator --no-assert -O3 -Wno-fatal --no-timing --trace --threads 1 --trace-underscore  -top-module $(PRJ)_tb $(VERILATOR_FLAGS) -cc ${HDL_FILES} --exe ${CPP_FILES} -o ../$(PRJ) -CFLAGS "${EXTRA_CFLAGS}" -LDFLAGS "${EXTRA_LDFLAGS}"
`