MiSTle-Dev/NanoMig #59: Reset by Key Combination (Ctrl+Amiga+Amiga)

Mangerrione, 2025-08-30:
When the Amiga user wants to reset the machine, he or she usually pushes down three keys on the keyboard:

**Ctrl+ Amiga+Amiga** on the Amiga Keyboard or
**Ctrl+Windows+Windows** in WinUAE

How could this behaviour be implementet in the NanoMig?
As it's Amiga specific it should be part of the core i guess?

luciust, 2025-09-01:
Hmmm - the ST(E) core in MySTery supports Ctrl+Alt+Delete

darius600, 2025-09-10:
https://github.com/MiSTle-Dev/FPGA-Companion/pull/98 would be a way to implement the Amiga soft reset via FPGA Companion. 

The MiSTeryNano core (opposed to NanoMig core) also simulates the keyboard controller (https://github.com/MiSTle-Dev/MiSTeryNano/blob/main/src/ikbd/ikbd.sv) which I’d expect handles the reset.

Since I just found it: https://github.com/MiSTer-devel/Main_MiSTer/blob/master/user_io.cpp#L3907 has an implementation for multiple cores.

Mangerrione, 2025-10-28:
Awsome!!! The "monkey grip" is a small detail but it's a huge improvement towards an authentic Amiga experience!