MiSTle-Dev/NanoMig #67: Tang Console with Mega 138k module

vanbako, 2025-09-19:
I have the Tang Console v1.1 with the 138k module and would like to add it to the project.
Can you tell me how to do this?

vossstef, 2025-09-19:
I will try over the next days try to add support for Console 138k with integrated onboard BL616 MCU. Is it that what you are looking for?  Likely i didn't catched fully your message.

harbaum, 2025-09-20:
As already stated per PM, we don't have these devices.

Also, you suggested asking nand2mario do the port. Did you contact him?

I'd really like to avoid doing things twice, and it seems you have now been asking many people to do the port for you.

vossstef, 2025-09-20:
@harbaum,  Apple2 is working fine on 138K PRo and Console 138k using onboard BL616.

After all it seems that many single issues:

- FPGA-companion timeout for detecting a core was too short for the almost 37 mbyte bitstream in gw5ast138k
- pll is still a mess. likely it is not the pllm IP block rather than ide sw function to create the code.
- latest 1.9.12 ide likely helped. maybe there had been corrections done. who knows 
- i steer the jtag to gpio switchover now under core control
- increased the bitstream flash load rate from 35mhz to 70mhz
- dedicated bl616 variant as auto switch ( ext spi / int bl616) does not work. root cause unknown. Maybe FPGA type specific. As we don't have errata sheet likely we will never find out.

harbaum, 2025-09-20:
My Mega138k does not run any of the latest cores in a usable manner. Sometimes the Atari ST seems to work for a few seconds, the NanoMig won't boot at all. I don't know if it's the cores or if I e.g. damaged the board as I bend it quite a lot when plugging the memory module which I only have one which is shared between the primer25k and the mega138k ...

vossstef, 2025-09-21:
@harbaum feeling that there are problems with the IDE PLL creation starting from 1.9.11 onwards including the 1.9.12 especially for the AST138k devices. 
I got the NanoMig working on a 138k Pro integrated onboard BL616. Some tcl / project attributes i had to adjust to compile it under 1.9.10.03. No way so far in 1.9.11 and 1.9.12... It simply does not work. Will push soon as extra variant

harbaum, 2025-09-21:
Yeah ... strange things happen. While the main system sometimes does work, the sd card often completely fails or hdmi would not work. Both could be explained by clock/pll problems. But these are by no means typical problems to see. Especially the sd card is a small isolated component that basically never exposes issues.

ellisgl, 2025-09-27:
@vossstef When I try to flash the kick start:
$ openFPGALoader --external-flash -o 0xc00000 ~/Desktop/Kickstart\ v3.1\ r40.063\ ShapeShifter.rom 
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file DONE
Jtag frequency : requested 2.50MHz    -> real 2.00MHz   
after program flash: displayReadReg 00004000
           [14] Security Final
        [24:23] CPU Bus Width: no BWD pattern is detected
Erase SRAM DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz   -> real 6.00MHz   
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash overflow
Error: write to flash failed
Error: Failed to program FPGA: Error: write to flash failed

harbaum, 2025-09-27:
What's a shape shifter Rom? Are you sure that's a kickstart ROM? What size is that file? A valid kickstart should be 256k or 512k.

ellisgl, 2025-09-27:
@harbaum https://archive.org/details/commodore-amiga-firmware

harbaum, 2025-09-27:
Also your flash chip seems to be only 64mbit/8mbytes in size. The mega138k I have has a 128mbit/16mbytes flash chip.

That would be a major problem since the cores for that bigger FPGA are bigger and occupy more space inside the flash rom leaving even less space for the retro ROMs like the kickstart.

harbaum, 2025-09-27:
The Tang Console 138k is supposed to be using the regular Tang Mega 138k which in turn should have a 128mbit flash rom according to:

https://wiki.sipeed.com/hardware/en/tang/tang-mega-138k/mega-138k.html

Can you take a closer look at your board? The flash chip is a small chip with 8 pins. It's supposed to be a 128mbit chip but the flash tool reports a 64mbit chip

ellisgl, 2025-09-27:
Yup, it's a WinBond 25Q64... I wonder if they updated them later on, or I got screwed some how. And I don't have the tools to swap that out, since it's smaller than an SOIC-8. I had to pull out the microscope.

harbaum, 2025-09-27:
I doubt it. The console is IMHO much newer than the mega 138k som on it. The mega 138k som was already used with the mega 138k dock. 

It looks to me like they just equipped some of their boards with a smaller flash. We could now try to re-arrange things a little bit. But this is just another undocumented Tang board variant we need to work around. The Tang Nano 20k also exists in various slightly different versions.

The solution for you would be to modify the core to expect the ROM to be located in the lower 8MB of the flash. It actually _is_ in the lower 8MB on most other boards. But that location is needed by the bigger FPGA. I hence moved the kickstart ROM location up on the mega 138k.

vossstef, 2025-09-27:
@harbaum , 
i will use identical address for Console 138k than TM60/ TC60k (C64, VIC20) :  0x700000 and guess we could use for Mig, MiSTery 0x500000. Guess i need to change the luckly dedicated Console 138k designs and re-compile...

ellisgl, 2025-09-28:
Yeah - I order a new one - just to see if it's been updated since. Strange how they ended up with a smaller flash, which definitely throws the specs off which throws off the devs. The 138K has been a mess.

harbaum, 2025-09-28:
You guys have too much money ...

vossstef, 2025-09-28:
@ellisgl @harbaum @darius600 i re-arranged the FLASH ROM location for all 138k boards.
It' is working on my 138k Pro.
@ellisgl once again some work... Could you please have a look into Console 138k ? Thx !

vossstef, 2025-09-29:
@ellisgl will check once again tomorrow. I realized that GW 1.9.12 Programner has issues with binary file programming and in addition address location entry changed (was extended).  You need to program at location 0x600000 an 1.3 KS

ellisgl, 2025-09-30:
@vossstef Clearing the flash, loading up the FS - black screen (used to be blue). Loading up a 512K rom at 0x600000 - black screen.

vossstef, 2025-09-30:
@ellisgl looks like once again that 138k with FLASH support and not having physical HW in hand will up endless pain...
Coukld you pls. have a look with Hex Editor and check that at offset 0x400000 the 2nd copy of 256kB KS 1.3 is present in your 512k file if you use that ?

I am still wondering why none of the two varaints are not working at all. Expected that one of it should do.

darius600, 2025-10-01:
Is there are way to get debug output from FPGA-Companion? I tried [this](https://wiki.sipeed.com/hardware/en/tang/tang-console/mega-console.html#The-system-does-not-recognize-the-onboard-debugger) but so far no luck.

Rafal85, 2025-10-01:
With nanomig_tc138k_bl616_01_10_25.fs the KS 1.3 or 3.1 logo appears for me, but then its stuck. I am new to this, maybe something not working with my sd card.

ellisgl, 2025-10-01:
Now I need to find a disk that works. My first pick quickly loaded back to the logo.

ellisgl, 2025-10-01:
OK - Found some stuff that works. Now,how do we pass the F keys to the apps?

ellisgl, 2025-10-07:
The new TC 139K came in, it too has a 64Mbit chip, not the 128... Thanks Sipeed for all the bad documentation.

darius600, 2025-10-21:
I build FPGA-Companion and Nanomig from source and it worked on TC138k. I was able to load Silkworm via Kickstart 1.3. 👌

My earlier USB issues were a user error: I had connected my USB devices to the USB-A ports of the TC instead of using a hub with PD on the USB-C port. This doc helped me to find the correct setup: https://github.com/MiSTle-Dev/.github/wiki/Board-Setup

Finally, I found flashing with the new standalone 1.9.12 SP1 Programmer more reliable than with the earlier tools I’ve tried.